UbiquityRobotics/ubiquity_motor #9: Investigate issue of robot going backward on startup

rohbotics, 2016-11-23:
@jim-v Can you add more detail of what is going on?

jim-v, 2016-11-23:
On kinetic, the wheels go backwards when running the motor node.   This behavior is new and started after an apt-get update.   The same hardware works when driven with indigo.

Added some debugging:

flycam@igloo1:~/catkin_ws/src/ubiquity_motor$ git diff
diff --git a/src/motor_hardware.cc b/src/motor_hardware.cc
index 0916969..9a37969 100644
diff --git a/src/motor_hardware.cc b/src/motor_hardware.cc
index 0916969..9a37969 100644
--- a/src/motor_hardware.cc
+++ b/src/motor_hardware.cc
@@ -228,9 +228,9 @@ void MotorHardware::writeSpeeds() {

     motor_serial_->transmitCommand(both);

-    // ROS_ERROR("velocity_command %f rad/s %f rad/s",
-    // joints_[0].velocity_command, joints_[1].velocity_command);
-    // ROS_ERROR("SPEEDS %d %d", left.getData(), right.getData());
+    ROS_ERROR("velocity_command %f rad/s %f rad/s",
+      joints_[0].velocity_command, joints_[1].velocity_command);
+    ROS_ERROR("SPEEDS %d %d", left_tics, right_tics);
 }

 void MotorHardware::requestVersion() {
@@ -361,4 +361,4 @@ int16_t MotorHardware::calculateTicsFromRadians(double radians) const {

 double MotorHardware::calculateRadiansFromTics(int16_t tics) const {
     return (tics * VELOCITY_READ_PER_SECOND / QTICS_PER_RADIAN);
-}
\ No newline at end of file
+}
diff --git a/src/motor_serial.cc b/src/motor_serial.cc
index d1853e2..dc68f6b 100644
--- a/src/motor_serial.cc
+++ b/src/motor_serial.cc
@@ -131,7 +131,7 @@ void MotorSerial::SerialThread() {
                 did_update = true;

                 RawMotorMessage out = getInputCommand().serialize();
-                ROS_DEBUG("out %02x %02x %02x %02x %02x %02x %02x %02x", out[0],
+                ROS_INFO("out %02x %02x %02x %02x %02x %02x %02x %02x", out[0],
                           out[1], out[2], out[3], out[4], out[5], out[6],

And the following output is produced (there is no /cmd_vel):

[INFO] [1479868831.674785]: Started controllers: ubiquity_velocity_controller, ubiquity_joint_publisher
[ INFO] [1479868831.676574043]: out 7e 3b 2a 00 00 00 00 9a
[ERROR] [1479868831.724382410]: velocity_command -0.248771 rad/s -0.736460 rad/s
[ERROR] [1479868831.724427372]: SPEEDS -2 -6
[ INFO] [1479868831.726547971]: out 7e 3b 2a ff fe ff fa a4
[ERROR] [1479868831.774328601]: velocity_command -0.497527 rad/s -1.472878 rad/s
[ERROR] [1479868831.774375113]: SPEEDS -4 -12
[ INFO] [1479868831.776539558]: out 7e 3b 2a ff fc ff f4 ac
[ERROR] [1479868831.824320219]: velocity_command -0.746286 rad/s -2.209303 rad/s
[ERROR] [1479868831.824366342]: SPEEDS -6 -18
[ INFO] [1479868831.826556813]: out 7e 3b 2a ff fa ff ee b4
[ERROR] [1479868831.874304304]: velocity_command -0.995013 rad/s -2.945633 rad/s
[ERROR] [1479868831.874364417]: SPEEDS -8 -24
[ INFO] [1479868831.876548685]: out 7e 3b 2a ff f8 ff e8 bc
[ERROR] [1479868831.924423884]: velocity_command -1.243876 rad/s -3.682367 rad/s
[ERROR] [1479868831.924555034]: SPEEDS -10 -30
[ INFO] [1479868831.926519381]: out 7e 3b 2a ff f6 ff e2 c4
[ERROR] [1479868831.974330743]: velocity_command -1.492555 rad/s -4.418555 rad/s
[ERROR] [1479868831.974385468]: SPEEDS -12 -36
[ INFO] [1479868831.976553581]: out 7e 3b 2a ff f4 ff dc cc
[ERROR] [1479868832.024319941]: velocity_command -1.741343 rad/s -5.155065 rad/s
[ERROR] [1479868832.024393351]: SPEEDS -14 -42
[ INFO] [1479868832.026537891]: out 7e 3b 2a ff f2 ff d6 d4
[ERROR] [1479868832.074321353]: velocity_command -1.990153 rad/s -5.891640 rad/s
[ERROR] [1479868832.074370837]: SPEEDS -16 -48

jim-v, 2016-11-23:
Committed debug changes to kinetic-debug branch

rohbotics, 2016-11-23:
Use a gist to hold code, it looks a lot nicer

https://gist.github.com/

rohbotics, 2016-11-23:
Can you try building ros-controllers from source?

https://github.com/ros-controls/ros_controllers

jim-v, 2016-11-23:
It still seems wrong built against kinetic-devel branch of ros_controllers:

[ERROR] [1479873691.189472100]: velocity_command -6.600985 rad/s -13.103448 rad/s
[ERROR] [1479873691.189518522]: SPEEDS -54 -107
[ INFO] [1479873691.193760741]: out 7e 3b 2a ff ca ff 95 3d

rohbotics, 2016-11-23:
Can you try adding debug prints near this line
https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/diff_drive_controller/src/diff_drive_controller.cpp#L312 ?

jim-v, 2016-11-23:
Added:

        ROS_ERROR("curr_cmd: %f %f", curr_cmd.lin, curr_cmd.ang);

Result:

    [ERROR] [1479877281.274923096]: curr_cmd: 0.000000 0.000000
    [ERROR] [1479877281.275014950]: velocity_command -6.600985 rad/s -13.103448 rad/s
    [ERROR] [1479877281.275050369]: SPEEDS -54 -107
    [ INFO] [1479877281.278006869]: out 7e 3b 2a ff ca ff 95 3d

jim-v, 2016-11-23:
Also, verified that curr_cmd does change as expected when `/teleop_twist_keyboard.py` is run.

rohbotics, 2016-11-23:
Something is acting really screwy between the diff_drive_controller and motor_hardware.

What are the warnings you are getting?

jim-v, 2016-11-23:
~~~~
Scanning dependencies of target ubiquity_motor
[ 28%] Building CXX object ubiquity_motor/CMakeFiles/ubiquity_motor.dir/src/motor_message.cc.o
[ 42%] Building CXX object ubiquity_motor/CMakeFiles/ubiquity_motor.dir/src/motor_hardware.cc.o
[ 57%] Building CXX object ubiquity_motor/CMakeFiles/ubiquity_motor.dir/src/motor_serial.cc.o
Generating reconfiguration files for PID in ubiquity_motor
Wrote header file in /home/flycam/catkin_ws/devel/include/ubiquity_motor/PIDConfig.h
[ 57%] Built target ubiquity_motor_gencfg
In file included from /opt/ros/kinetic/include/hardware_interface/robot_hw.h:35:0,
                 from /home/flycam/catkin_ws/src/ubiquity_motor/include/ubiquity_motor/motor_hardware.h:36,
                 from /home/flycam/catkin_ws/src/ubiquity_motor/src/motor_hardware.cc:30:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static char (& hardware_interface::CheckIsResourceManager<T>::callCM(std::vector<C*>&, C*, typename C::resource_manager_type*))[1]’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:73:3: warning: no return statement in function returning non-void [-Wreturn-type]
   }
   ^
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static char (& hardware_interface::CheckIsResourceManager<T>::callCM(std::vector<C*>&, C*, ...))[2]’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:77:74: warning: no return statement in function returning non-void [-Wreturn-type]
   static no& callCM(typename std::vector<C*>& managers, C* result, ...) {}
                                                                          ^
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static std::vector<std::__cxx11::basic_string<char> > hardware_interface::CheckIsResourceManager<T>::callGR(T*, ...)’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:93:58: warning: no return statement in function returning non-void [-Wreturn-type]
   static std::vector<std::string> callGR(T* iface, ...) {}
                                                          ^
[ 71%] Linking CXX shared library /home/flycam/catkin_ws/devel/lib/libubiquity_motor.so
[ 71%] Built target ubiquity_motor
Scanning dependencies of target motor_node
[ 85%] Building CXX object ubiquity_motor/CMakeFiles/motor_node.dir/src/motor_node.cc.o
In file included from /opt/ros/kinetic/include/hardware_interface/robot_hw.h:35:0,
                 from /home/flycam/catkin_ws/src/ubiquity_motor/include/ubiquity_motor/motor_hardware.h:36,
                 from /home/flycam/catkin_ws/src/ubiquity_motor/src/motor_node.cc:35:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static char (& hardware_interface::CheckIsResourceManager<T>::callCM(std::vector<C*>&, C*, typename C::resource_manager_type*))[1]’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:73:3: warning: no return statement in function returning non-void [-Wreturn-type]
   }
   ^
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static char (& hardware_interface::CheckIsResourceManager<T>::callCM(std::vector<C*>&, C*, ...))[2]’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:77:74: warning: no return statement in function returning non-void [-Wreturn-type]
   static no& callCM(typename std::vector<C*>& managers, C* result, ...) {}
                                                                          ^
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In static member function ‘static std::vector<std::__cxx11::basic_string<char> > hardware_interface::CheckIsResourceManager<T>::callGR(T*, ...)’:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:93:58: warning: no return statement in function returning non-void [-Wreturn-type]
   static std::vector<std::string> callGR(T* iface, ...) {}
                                                          ^
[100%] Linking CXX executable /home/flycam/catkin_ws/devel/lib/ubiquity_motor/motor_node
~~~~

jim-v, 2016-11-28:
The problem occurs on:
Linux igloo1 4.4.0-47-generic #68-Ubuntu SMP Wed Oct 26 19:39:52 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
But not on:
Linux fxpi301 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l armv7l armv7l GNU/Linux

rohbotics, 2016-12-31:
I can't replicate this issue on my magni,
it could have been fixed upstream.

closed 2017-05-02 by jim-v