UbiquityRobotics/ubiquity_motor #34: Script to read Robot ID

rohbotics, 2018-01-15:
https://github.com/UbiquityRobotics/ubiquity_main/issues/84 

### Spec

If robot is detect as magni or loki, set the return code to 0, and print `magni` or `loki` to stdout.

If no robot detected set return code to 1 and exit. 
If unknown robot detected set return code to 2 and print the id (ASCII decimal) to stdout.

All other output (debugging/warnings/errors) must be sent on stderr.

@waynegramlich @jim-v Does that look reasonable?

I can implement it.

waynegramlich, 2018-01-15:
That will work, but I think we should be thinking ahead.

I would propose that it be done with command line options:
* `-a` `--all`: Print out all options.
* `-r` `--robot`: Print out robot platform (e.g. `loki`, `magni`, `none`)
* `-f` `--firmware`: Print out firmware revision
* `-p` `--pcb`: Print out the PCB revision
* `v`` `--verbose`: Print out debugging information.
It is not necessary to figure out all of the options right now. What we do want to do is lay the groundwork for additional options as we need them.

rohbotics, 2018-01-15:
Those seem like good forward-thinking additions. 

I would still have a non-zero return code for robot platform being `none`.
That way a bash script can do something like:
```
if ! robot=$(robot_info_script --robot); then
      # Do stuff
      roslaunch $robot foo
else
      # No robot so do nothing
fi
```

waynegramlich, 2018-01-15:
Agreed.

closed 2018-01-19 by rohbotics