UbiquityRobotics/ubiquity_motor #38: Motor Node error on Raspberry Pi3B+

anfederman, 2018-05-29:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Firmware not reporting its version
[motor_node-3] process has died [pid 3126, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/ubiquity_motor/motor_node /ubiquity_velocity_controller/cmd_vel:=/cmd_vel /ubiquity_velocity_controller/odom:=/odom __name:=motor_node __log:=/home/ubuntu/.ros/log/88bca176-6387-11e8-87fc-994ec5fe8335/motor_node-3.log].
log file: /home/ubuntu/.ros/log/88bca176-6387-11e8-87fc-994ec5fe8335/motor_node-3*.log

after running apt update upgrade,  and building ubiquity_motor in catkin_ws

I looked for the log files but they don't get written.

Robot runs on Pi 3B with no errors.

jim-v, 2018-05-30:
It may be trying to communicate over the wrong serial device.

Can you send the output of `lsusb` and `ls /dev/tty*`

rohbotics, 2018-06-02:
Any update on this Alan?

closed 2018-06-04 by rohbotics