UbiquityRobotics/ubiquity_motor #40: DESERIALIZATION ERROR! - 2

chrissunny94, 2018-05-30:
`terminate called after throwing an instance of 'serial::SerialException'
  what():  SerialException device reports readiness to read but returned no data (device disconnected?) failed.`

chrissunny94, 2018-05-30:
```
[ERROR] [1527686226.354403508]: Unknown Error
terminate called after throwing an instance of 'serial::SerialException'
  what():  SerialException device reports readiness to read but returned no data (device disconnected?) failed.
[motor_node-3] process has died [pid 4868, exit code -6, cmd /home/chris/ubiquity_ws/devel/lib/ubiquity_motor/motor_node /ubiquity_velocity_controller/cmd_vel:=/cmd_vel /ubiquity_velocity_controller/odom:=/odom __name:=motor_node __log:=/home/chris/.ros/log/abd30d56-640b-11e8-b418-5c93a2b68b2f/motor_node-3.log].
log file: /home/chris/.ros/log/abd30d56-640b-11e8-b418-5c93a2b68b2f/motor_node-3*.log

```

chrissunny94, 2018-05-30:
```
terminate called after throwing an instance of 'serial::IOException'
  what():  IO Exception (13): Permission denied, file /tmp/binarydeb/ros-kinetic-serial-1.2.1/src/impl/unix.cc, line 151.
```

This problem persists even after doing 

`sudo chmod 777 /dev/ttyS0`

The output for 

`ls /dev/tty*` 

```
/dev/tty    /dev/tty15  /dev/tty22  /dev/tty3   /dev/tty37  /dev/tty44  /dev/tty51  /dev/tty59  /dev/tty9
/dev/tty0   /dev/tty16  /dev/tty23  /dev/tty30  /dev/tty38  /dev/tty45  /dev/tty52  /dev/tty6   /dev/ttyAMA0
/dev/tty1   /dev/tty17  /dev/tty24  /dev/tty31  /dev/tty39  /dev/tty46  /dev/tty53  /dev/tty60  /dev/ttyprintk
/dev/tty10  /dev/tty18  /dev/tty25  /dev/tty32  /dev/tty4   /dev/tty47  /dev/tty54  /dev/tty61  /dev/ttyS0
/dev/tty11  /dev/tty19  /dev/tty26  /dev/tty33  /dev/tty40  /dev/tty48  /dev/tty55  /dev/tty62
/dev/tty12  /dev/tty2   /dev/tty27  /dev/tty34  /dev/tty41  /dev/tty49  /dev/tty56  /dev/tty63
/dev/tty13  /dev/tty20  /dev/tty28  /dev/tty35  /dev/tty42  /dev/tty5   /dev/tty57  /dev/tty7
/dev/tty14  /dev/tty21  /dev/tty29  /dev/tty36  /dev/tty43  /dev/tty50  /dev/tty58  /dev/tty8

```

jim-v, 2018-05-30:
I'm not sure what hardware you have, but the device that is connected to the motor controller is more likely to be `/dev/ttyAMA0`.  As for the `chmod`, read and write permission (666) should be sufficient.  It's not harmful to make a device executable, but it has no effect.

chrissunny94, 2018-05-31:
I changed it to 
`/dev/ttyAMA0`

gave the necessary permission , `sudo chmod 666 /dev/ttyAM0`

Now i get  `terminate called after throwing an instance of 'std::runtime_error'
  what():  Firmware not reporting its version
`

rohbotics, 2018-05-31:
How are you launching?
Is this on a Raspberry Pi running our image?

Is the Pi plugged in to the motor controller, and do you see leds on the
motor controller?

Hope to get this resolved soon
Rohan

chrissunny94, 2018-05-31:
I am launching this after sshing into the RPI . 
Nope i am not using the RPI image from the website , i compiled the code after pulling it off from github .

chrissunny94, 2018-05-31:
Problem solved after I used the ubiquity image , thanks .

closed 2018-05-31 by chrissunny94