UbiquityRobotics/ubiquity_motor #41: left/right PWM limit reached

richardw347, 2018-07-27:
We just received our first Magni (thanks guys!) and I'm just running some tests just connecting my laptop via FTDI and running the ubiquity_motor driver and sending some teleop commands. The robot is up on blocks at the moment (for safety). I'm just having an issue when I send velocity commands above 1 m/s. 

I've plotted commanded linear velocity vs reported to demonstrate the issue in this example the commanded linear velocity is 1.5m/s:

[image]

The robot reaches 1.1m/s but throws a of pwm limit messages:
```
[/motor_node ERROR 1532695069.064819969]: left PWM limit reached
[/motor_node ERROR 1532695069.064871139]: right PWM limit reached
```
Also as the graph shows setting the velocity set-point back to zero instead of immediately starting to decelerate the robot holds speed for a couple of seconds. 

Here is my base.yaml file, I'm using default config file from magni_robot repository I just changed the serial port address:
```
ubiquity_motor:
  serial_port: "/dev/ttyACM0"
  serial_baud: 38400
  controller_loop_rate: 20
  serial_loop_rate: 200
  pid_proportional: 5000
  pid_integral: 0
  pid_derivative: 0
  pid_denominator: 1000

ubiquity_joint_publisher:
  type: "joint_state_controller/JointStateController"
  publish_rate: 50

ubiquity_velocity_controller:
  type: "diff_drive_controller/DiffDriveController"
  left_wheel: 'left_wheel_joint'
  right_wheel: 'right_wheel_joint'
  publish_rate: 50
  pose_covariance_diagonal: [0.2, 0.2, 0.2, 0.2, 0.2, 0.2]
  twist_covariance_diagonal: [0.2, 0.2, 0.2, 0.2, 0.2, 0.2]
  cmd_vel_timeout: 0.25

  enable_odom_tf: true

  wheel_separation : 0.33
  wheel_radius : 0.1015

  base_frame_id: base_link

  # Wheel separation and radius multipliers
  wheel_separation_multiplier: 1.0 # default: 1.0
  wheel_radius_multiplier    : 1.0 # default: 1.0

  # Velocity and acceleration limits
  # Whenever a min_* is unspecified, default to -max_*
  linear:
    x:
      has_velocity_limits    : false
      max_velocity           : 1.0   # m/s
      has_acceleration_limits: true
      max_acceleration       : 0.5   # m/s^2
  angular:
    z:
      has_velocity_limits    : false
      max_velocity           : 2.0   # rad/s
      has_acceleration_limits: true
      max_acceleration       : 5.0   # rad/s^2

```
So in summary two issues:

- How to rectify the pwm limit issues for travelling at higher velocities.
- The error message seems to saturate the serial bus meaning subsequent velocity commands (e.g. to stop) don't get though for several seconds. 

Thanks!

mjstn, 2020-08-15:
The issue here  has been fixed with the ability now to set a ROS parameter of fw_max_pwm.
I was reviewing older issues and noticed this issue.    
I cannot say at what point this did or will enter our default images but it is in the ubiquity_motor repository at this time.
I feel it will likely be included in our default Magni images upon the next image released after todays date.

mjstn, 2020-08-15:
Closing this as it is felt to be fixed with recent inclusion of ROS paramater of fw_max_pwm.

closed 2020-08-15 by mjstn