UbiquityRobotics/ubiquity_motor #42: Motors moving on E-Stop release

richardw347, 2018-07-27:
I'm just testing the E-Stop on the Magni and noticed an issue with the wheels moving after the E-Stop is released. The procedure to repeat the issue is:

- Send a forward velocity command (I was sending 0.4m/s)
- Press the E-Stop
- Stop sending velocity commands
- Release the E-Stop

I made a short video demonstrating the issue: [https://youtu.be/hZxYnKrlZ9E](https://youtu.be/hZxYnKrlZ9E)

davecrawley, 2018-07-27:
We are prioritizing a firmware fix for this. Expect it patched in few weeks.

albuckley, 2018-07-27:
I am concerned about this issue from a functional safety point of view. Is the fix going to be 61508 compliant? When an e-stop is triggered there aught not to be any complex logic involved. It would appear that there is some persistence of memory involved. If your patch involves reworking faulty logic, I can't see it being compliant to functional safety requirements. The system should be totally wiped out following an e-stop event.

I don't recall off hand but I do believe functional safety standards outline handling of error events and specify procedures for recovery. This e-stop release procedure being one cited. It is a common problem with a defined solution. I don't think the solution is one of patching firmware to be compliant. The reason being, the system needs to be safe and protected against future patches that may potentially be buggy.

I would recommend a full system reset, forcing the user/ host system into following the initialization sequence before anymore motion commands accepted or acted upon.

An e-stop event is one that in practice should never occur. If such an event occurs it means all higher tier safety systems were compromised. Robotics engineers aim to make their sensor and avoidance systems highly robust and build in multiple diverse and redundant protection means. Realistically, these can all fail, and do more so than engineers would like. But when that does happen, an arduous recovery sequence is not a problem. In fact, it is desired in order to mitigate this type hazardous fault.

closed 2019-04-07 by rohbotics