UbiquityRobotics/ubiquity_motor #49: Battery Power Supply Status for Magni Robot

Vuwij, 2018-11-10:
https://forum.ubiquityrobotics.com/t/battery-power-supply-status-for-magni-robot/196

UbiquityRobotics/ubiquity_motor/blob/c084edfc796ed05224765634dd18618d019014c9/src/motor_hardware.cc

Line 154 shows that the battery power supply status is always marked as unknown.
It would be very helpful if it is possible to indicate the battery power supply status of the robot. This can be especially helpful for automating robots and putting a safety for when the robot is charged or not. If we were to use the robot in a delivery scenario for example. It is also helpful for a UI to know if the robots are being charged or not.

mjstn, 2020-08-15:
We cannot currently know if the robot is being charged.  We do not have a directional current sense between the XLR jack and the battery.  Voltage alone cannot tell us if we are charging although it can tell is if we are hooked to a charger and fully charged we cannot tell if we are in some other state of charging where we are not full yet.

mjstn, 2021-02-16:
This is a feature not possible with our hardware or software.  We will be pushing this feature to our next generation robot platform that is in development now.

closed 2021-02-16 by mjstn