UbiquityRobotics/ubiquity_motor #65: 6-State Odometery

rohbotics, 2019-04-23:
Use both rising and falling edges

rohbotics, 2019-06-15:
We need a test procedure for this.

mjstn, 2019-06-16:
Minimal Test Procedure:   
- Get v35 firmware from Mark or wait till it becomes available internally for firmware uploader.
- Clone ubiquity_motor repository and do catkin_make on raspberry Pi side. (REQUIRED!)
- As a minimum do the tests up to ESTOP testing on https://learn.ubiquityrobotics.com/verification
- Do other movement based tests with whatever other approaches you tend to use so we try as many things as possible as everyone does things a little different and looks at the robot in a different way.

mjstn, 2020-02-03:
To use 6-state firmware you must have host code and v35 firmware.

**Host Code:  Get this first because it works with older OR new v35 firmware**
cd ~/catkin_ws/src
git clone https://github.com/UbiquityRobotics/ubiquity_motor
cd ~/catkin_ws
catkin_make -j 1   

Get v35 Firmware on the main MCB board
Use these instructions and specify  v35 for the version
https://learn.ubiquityrobotics.com/firmware-upgrade

Mark

mmaraja, 2020-02-07:
This is the list of what the robot should be able to do:

1.Try putting the weight of 100kg on the robot. It should go with the same ease as before.
- The robot is capable of caring the 100kg on itself and there shoud not be a change in it's movement.
2. Try driving the robot with it's maximum speed, which is 1m/s.
- The robot has the maximum speed of 1m/s (it is software limited).
3. Drive the robot over the bump that is 0.75".
- The robot should be able to go over a bump easily.
4. Climb it at 4 different angles to see it's maximum potential.
- The robot shoud be able to climb at a certain angle without issue.
5. Place an 0.5" object on the floor and command it left/right to go over it in turn. It should be able to go over it easily and its course should not be disrupt.
- On the left/right command it has to be flexible and be able to go over minor obstacle.
6. Try to drive it on epoxy floor and carpet and see the flexibilty, it should be working on any surfaces that are stable, firm, and slip resistant. 
- Openings in ground and floor surfaces, such as grates, are limited in width of a 3" rare wheel of the robot.
- The robot will work on carpet without any issues if the standards specify the maximum pile height (1/2” measured to the backing, cushion, or pad). Carpeting must be securely attached so that it does not shift or buckle once robot is on it. Cushions or pads, if used, also must be properly secured to resist movement.
7. Try driving it over a change in levels (it should not exceed 1/4" in height). 
- It should go over a change of 1/4" without any issues. 
8. Try driving it on a ramp. Ramps should be for heights over the 1/4", when you can't drive over a change in level easily.
- Ramps and curb ramps are required along accessible routes to span changes in level greater than ½”.

closed 2020-02-15 by rohbotics