UbiquityRobotics/ubiquity_motor #79: Firmware instructions reference missing upgrade_firmware.py

MoffKalast, 2020-02-12:
Firmware instructions at https://learn.ubiquityrobotics.com/firmware-upgrade reference missing script upgrade_firmware.py as it isn't packaged along with the compiled code.

The fix can be to either change the instructions to a git clone of the repository that has the script or to package the python scripts along with the compiled code.

rohbotics, 2020-02-12:
@MoffKalast I just checked, and it looks like the script is being packaged as part of the deb. Can you explain what the error you encountered is?

MoffKalast, 2020-02-12:
That does not seem to be the case on the magni I'm working on right now.

```
ubuntu@magni2:~$ roscd ubiquity_motor
ubuntu@magni2:/opt/ros/kinetic/share/ubiquity_motor$ ls
cmake  package.xml
ubuntu@magni2:/opt/ros/kinetic/share/ubiquity_motor$ ls cmake
ubiquity_motorConfig-version.cmake  ubiquity_motorConfig.cmake
```

rohbotics, 2020-02-12:
roscd doesn't really work for binary packages, as all the components are
placed in different places.

What happens when you run `rosrun ubiquity_motor upgrade_firmware.py` ?

Rohan

MoffKalast, 2020-02-12:
Well it looks in that very directory for some reason.

`[rosrun] Couldn't find executable named upgrade_firmware.py below /opt/ros/kinetic/share/ubiquity_motor`

rohbotics, 2020-02-19:
Hey @MoffKalast 

I just tried this again on the 2020-02-04 image, and it `rosrun ubiquity_motor upgrade_firmware.py` runs perfectly fine.

closed 2020-02-19 by rohbotics

MoffKalast, 2020-02-19:
¯\\\_(ツ)_/¯ May have something to do with removing and reinstalling packages then. I'll look into it more sometime.