UbiquityRobotics/ubiquity_motor #80: Changes to the motor node for the default image

mmaraja, 2020-02-14:
Changed values of parameters for motor node.
Improved motion and reduced deviations. 
PID_P: 5000
PID_I: 7
PID_D: -110
PID_C: 1000
PID_W: 70
PID_V: 15000

rohbotics, 2020-02-14:
Hey @mmaraja (and @davecrawley),

Are these parameters known to be good on both 6 state and non 6 state robots? Do we need a method to change the loaded PID params based on the type of motor contoller?

davecrawley, 2020-02-14:
These parameters haven't been tested on a non-six state robot.

That said the parameters are very close to the parameters we were putting
on our non-six state robots. The key difference is the addition of the V
term - which is not available on older firmware revisions. There are a few
other subtle changes as well (for example the windowing parameter was
narrowed to optimize performance).

David

rohbotics, 2020-02-16:
Ok, we really should test on a "v32" robot before we can make these changes the default.

rohbotics, 2020-02-20:
I created a PR for this, but we still need to test on a v32 robot.

mmaraja, 2020-02-21:
The test on a v32 robot is done, new values of parameters can be default.

closed 2020-02-21 by rohbotics