UbiquityRobotics/ubiquity_motor #96: motor_node process has died

Alex-Beh, 2020-07-29:
Hello, I face the following issue when I `roslaunch magni_bringup core.launch`

> symbol lookup error: /home/followme/follow_me/devel_isolated/ubiquity_motor/lib/libubiquity_motor.so: undefined symbol: _ZN6serial6SerialC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEjNS_7TimeoutENS_10bytesize_tENS_8parity_tENS_10stopbits_tENS_13flowcontrol_tE

Besides, when I run python test_motor_board.py and it raise the following error:

> NOTE: Program requires prior use of:  sudo systemctl stop magni-base

However, when I run the command `sudo systemctl stop magni-base`, it said:

> Failed to stop magni-base.service: Unit magni-base.service not loaded.

I am using magni robot connect to nvidia xavier nx using ftdi connector.

rohbotics, 2020-07-30:
Hi @Alex-Beh, If you are not using our image, you don't have to worry about the systemctl stop command.

How did you compile and run ubiquity_motor on your platform? It looks like you are missing a library. 

What version of ROS are you running?

mjstn, 2021-02-16:
I am retiring this issue for lack of repeatability on current production image.  This seems to be related to use of the motor node on a system other than a Magni.

closed 2021-02-16 by mjstn