UbiquityRobotics/ubiquity_motor #121: Need motor stop command for collision stop on system_control topic

mjstn, 2021-02-15:
There is a need for friendlier high level interface to stop motor control for cases such as collision.
This came in through a forum user who feels very strongly this should exist.
Because it is a request for a safety related feature to make collision safety easier it seems a good idea.

We will use the /system_control topic recently added for motor disable for firmware upgrade.
We could have motor_node which listens now on this topic as well as /cmd_vel be the control point for this feature

MoffKalast, 2021-02-15:
Why don't we just use the already implemented enable/disable functionality? It should get the job done, unless it takes too long to break/remake the serial connection.

mjstn, 2021-02-15:
No, that is 'harsh' and totally re-initializes MCB.  We want to disable speed control only for collision safety reasons.

mjstn, 2021-02-16:
This ability as well as some cleanup to encapsulate this and the mcb disable into the motor_node class for state is all now done.
This is located in ubiquity_motor branch of   addSpeedControlForMotorNode

mjstn, 2021-04-13:
This has been implemented in ubiquity_motor in commit f130510bf23d5c38916ef82f955a91d4068d018c in kinetic-devel

This has also been used in a new node called sonar_detector that acts as an independent collision detect mechanism that can stop the motor node by use of this new topic.

closed 2021-04-13 by mjstn