UbiquityRobotics/ubiquity_motor #127: Offset battery state so 50% reads as 0%

MoffKalast, 2021-04-26:
Something we talked about the other day, but then forgot to turn into an issue.

To sum up, the RPi becomes unstable under what's currently being calculated as 50%, so we should probably label it as far lower, like 5% or 10% at most so it's clearer that the robot is about to stop working.

I suppose the question is what do we do with the remaining 0-49%. We could just have it linearly interpolate between 0-5% or something or just have it display as zero. Anyhow, up to the guy who implements it as it's more up to personal preference.

mjstn, 2021-05-07:
We don't want to let our batteries operate much lower than around 12V which is near that 50% capacity.
We are unable to monitor anything but battery voltage.   Some tweeks have gone into the MCB that apply to better tolerance and value resistors used to measure voltage as well as a fix for the ratio so older boards were often way off in battery readings.
 
Many battery mfgs have lots of info on the nature of battery life as a function of how far you regularly discharge the battery.  This one seems rather good coverage in terms of how a battery overall life is shortened by a given regular depth of discharge (DOD).
http://www.batteriesinaflash.com/deep-cycle-battery-faq

MoffKalast, 2021-05-07:
I mean according to some of our customers, there's a pretty high chance of sd card corruption below 50%, so I'd say we should set a hard limit there and not even debate deep cycles. I'd imagine the 5V rail must become unstable somehow but we don't exactly have proper tests of it.

mjstn, 2021-05-07:
Totally agree.   I was trying to say, stop at 50% discharge which is in the 'area' of 12V
I have as of last week completed an enhancement to my battery load pulser and was hoping to do more 'exact tests' but frankly we should take advice from the hundreds of folks that have used gel cells before and trust their judgement in order to not 'push our luck' and corrupt a lot more SD cards simply to get an extra 30 minutes of robot life.

MoffKalast, 2021-05-07:
This has me thinking though if we have two 7.2Ah batteries in series that gives us what, like 4Ah@28V to use in total?

I think there are 20V 6Ah cordless drill batteries out there that might get us similar range and we could probably fit like four of them for the same space/mass. Sounds totally ridiculous but it could be something worth trying out.

PR #138 (2021-08-30): Battery percentage calculator
This PR should address the issue that one of our clients has raised and I've noticed as well: full charge voltage level is off by about 30%, and the max possible achievable percentage is like 77% which is odd for just about everyone. From what I've tested there's no way our usual 7.2Ah 12S lead acids can go above ~26.1V when not plugged into a charger.

Aside from a fix for that I've also ported my usual battery calculation system, which uses a lookup table along with level interpolation and should be much more accurate than the simple interpolation we had before. [... cut]