UbiquityRobotics/ubiquity_motor #129: Need better resolution on motor speed in order to have better app control of path

mjstn, 2021-05-12:
A customer has complained that his app cannot control a path smoothly on a Magni and I have determined this is because the resolution of the speed control bits we send the MCB is too course to allow smooth control of a path by an app or nav package.  An app is not able to control very gentle rotations on the order of under 2 degrees per second with todays firmware.

If we wanted to solve this issue a significant task to take on would be to allow the speed values sent to the MCB to be on the order of integers that were 10x or more in value for the same given end speeds we use.

Today the MCB firmware is sent integer values for speed control. A single count for speed control is on the order of 0.02 meters per second. (roughly). At a linear speed of 0.14M/Sec the motor controller gets a count of about 12 for right and left motor. If you set angular (using e and c keys) to a value of around what you found of 0.025 range then one motor can get a count of 12 and the other a count of 11.

To resolve this issue we would have to have a new speed command only used if firmware were above some given version.  This new speed message still would use 2 packed 16-bit integer values but the scale of units would be such that a value of lets say 16 times larger would be used for any given speed.  So for this example of 0.14M/Sec today using a setting of 12 the new speed control message would use something like 192 and in that way minimal resolution on angular rotation would improve 16x

tp4348, 2021-05-12:
I think you made a great point. One thing for breadcrumb is that minimal linear and angular velocities depend also upon the payload that is on the robot. 

I would propose for Gen6 to have some kind of **mass/force sensors** such that we could determine the payload on the robot and have our configuration parameters dependent upon that. This would indeed come in useful maneuvering since the Motor Controller PIDVs need to be different and minimal velocities in move_basic need to be updated etc. It might be only motor controller PIDVs, just that then on the minimal velocity from move_basic the motor controller would react more sharply (due to different PIDVs).

MoffKalast, 2021-05-12:
I mean, it's definitely something to implement but I have some doubts regarding what it would actually fix. It's not like our localization is good enough for this to be the bottleneck when trying to move around accurately.

mjstn, 2021-05-12:
Thanks for the inputs so far. I mostly wanted to make all 3 'team leads' be aware of this because I think it has sort of come up but not really had root cause be identified.    Aisle mode also 'waddles' along with obvious corrections slowly happening and that too may be due to this granularity.

mjstn, 2021-09-22:
I feel this is best pushed out to Gen6 because it would have significant risk of breaking things (badly) and it has very little real benefit right now.     

closing this now so we remain 'realistic' in what we try to improve on current platform of Magni

closed 2021-09-22 by mjstn