UbiquityRobotics/ubiquity_motor #137: Breadcrumb behaving incorrectly on firmware v41

mbostic, 2021-07-06:
I was trying v41 firmware on Breadcrumb, but it didn't work as it should. The most noticeable thing was that the robot was moving very slowly. It was also turning in a bit wrong directions, so I guess that also odometry didn't work perfectly.

tp4348, 2021-07-19:
We need more information on that. Did that happened after we drove the robot with 100kg, is robot underpowered(low battery), are we able to reproduce that also with keyboard_teleop.

mbostic, 2021-07-23:
It happens regardles of the load before testing. The robot wasn't underpowered. It also happens with keyboard_teleop (the slow driving part).
Here is the video of the problem. Instantly after flashing back the v40 firmware, it works normally.
https://workdrive.zoho.com/file/opxqf0ec321edd7e94b9291f801b82efd9cc8

mjstn, 2021-09-18:
The slow driving is because that is a firmware made with all defaults for the MoonStar product.
I am closing this now because v43 has been successfully tested.
Please refer to issue:  https://github.com/UbiquityRobotics/robot_testing/issues/12
and Pull request: that has been merged to ubiquity_motor was called merge4wdFeaturesToMagni

closed 2021-09-18 by mjstn