UbiquityRobotics/ubiquity_motor #150: MCB HARDWARE: Make 5V main less likely to reboot the Pi4 or cause Ethernet dropout

mjstn, 2021-08-24:
It seems we need to revisit the power supply for 5V main for reliability reasons.

These items will be evaluated and some or all of them will be decided based on results.

- R1314 changes to 1.13k 1% for a 5.15V supply
- D2303 and D3303 become DIODE ZENER 5.6V 5W DO214AA
- C1309, C1310, C1311 revert back to 100uf ceramic (we had to change these for production parts shortages)
- Perhaps C1312 should be much larger than 1000 uF 16V.  Perhaps change to 10V as high as we can find capacitance

mjstn, 2021-09-22:
We are being more conservative and are to use a value of 1.15k for R1314 for a 5.10V supply.

We are completely removing D2303 and D3303 because they are not really needed except to prevent some catastrophic failure of the buck converters for 5V but if that happens the MCB is badly broken in any case anyway.

We are as of 9-20-2021 doing rework of all in-stock MCB boards for above items.

The last items on capacitors are still to be evaluated for future MCB versions

mjstn, 2021-12-15:
Closing this issue today.   Status:   We have made 100 boards with above mod and they appear working.  We will ship this starting with Magni units on rev 5.3 MCB boards with the MOD 5.1 on them.
We have also gotten our PC group to be updating the BOM for the MCB so that the next run will be with these changes.

closed 2021-12-15 by mjstn

MoffKalast, 2021-12-16:
Good to know, thanks for the update 👍