UbiquityRobotics/ubiquity_motor #167: upgrade_firmware.py syntax error

JanezCim, 2022-03-14:
Not entirely unexpected, the over the web MCB firmware upgrade fails in a string error (likely yet another phthon thang)   Here is the failure:
```
ubuntu@pi-focal:~$ rosrun ubiquity_motor upgrade_firmware.py
-------------------------------------------------------------
Welcome to the Ubiquity Robotics Firmware Updater!

Please make sure that you are not running any ROS nodes.
- sudo systemctl stop magni-base

Note: Updating the firmware requires access to the internet.
-------------------------------------------------------------

Please enter your email address: mjstn2011@gmail.com
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/ubiquity_motor/upgrade_firmware.py", line 43, in <module>
    email = input("Please enter your email address: ").strip()
  File "<string>", line 1
    mjstn2011@gmail.com
             ^
SyntaxError: invalid syntax

```

_Originally posted by @mjstn in https://github.com/UbiquityRobotics/robot_testing/issues/27#issuecomment-1065961144_

JanezCim, 2022-03-14:
Trying to solve in 6086adb0909ddde8a574fa93384201e983de829a but currently i got 500 server error:
```
ubuntu@pi-focal:~/catkin_ws/src/ubiquity_motor$ rosrun ubiquity_motor upgrade_firmware.py 
-------------------------------------------------------------
Welcome to the Ubiquity Robotics Firmware Updater!

Please make sure that you are not running any ROS nodes.
- sudo systemctl stop magni-base

Note: Updating the firmware requires access to the internet.
-------------------------------------------------------------

Please enter your email address: j.cimko@gmail.com
j.cimko@gmail.com
Error: 500 Internal Server Error. Something went wrong, try again in a few minutes. If the error persists, contact support.

```

JanezCim, 2022-03-14:
I've also tried this with the unchanged code from kinetic-devel but on focal OS with python2 installed and got the same error. Could this be a server side error @rohbotics ?

anfederman, 2022-03-14:
HMM... I saw the same thing, and thought it was a python thing.  I used a
kinetic image to upgrade to firmware 40, but did locally. When I tried oon
Noetic I got the sting/error/server I don't recall if I got a token.

JanezCim, 2022-03-17:
syntax error resolved with resolving the https://github.com/UbiquityRobotics/ubiquity_buildbot/issues/9

closed 2022-03-19 by JanezCim