UbiquityRobotics/ubiquity_motor #172: Controller Spawner error

WafaAbid1, 2023-05-25:
After launch magni_bringup core.launch no error or warning appears but I cant have the robot's model on Rviz. When shutting the node I have these warnings:  Controller Spawner error while taking down controllers: 

unable to connect to service: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details. 

and [WARN] [1685020056.103366]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer.

Any ideas on how to fix it? It's always happening

MoffKalast, 2023-05-25:
For the robot model iirc you need to clone magni_robot into a local repo, maybe launching magni_description on the workstation itself, but but that may have some conflicts with something. 

I think the controller spawner error is unrelated and likely not causing any issues, probably something to do with killing the diff drive controller.

anfederman, 2023-05-25:
Are you communicating with the MCB?  Can you get the battery status with
rostopic echo ?  Also If your robot has the OLED, is it displaying battery
voltage?
If OLED is ok but not getting voltage to the workstation, this might be a
network/wifi issue.